bithollow/bitforge #7: [M1][High][HW] Solder PCA9685 prototype on breadboard

droneboost, 2015-04-15:


muzi-15, 2015-04-16:
the total cost this time is 110 + 14 + 23 = 147, the detailed information is as follows:
1. pca9685 chip x5  22X5
    http://detail.tmall.com/item.htm?id=41866364083&spm=a1z09.2.9.10.rCck0g&_u=g35sn0n834d&mt=
2. resistance 7X2
    http://detail.tmall.com/item.htm?  id=39813854420&spm=a1z09.2.9.27.rCck0g&_u=g35sn0nf703&mt=
3. convert board and bread board  23
    http://detail.tmall.com/item.htm?id=25072296568&spm=a1z09.2.9.44.rCck0g&_u=g35sn0n791e&mt=
   http://detail.tmall.com/item.htm?id=25282796860&spm=a1z09.2.9.64.rCck0g&_u=g35sn0n12aa&mt=

onceme, 2015-04-16:
How much should I pay you? 147/3=49?

droneboost, 2015-04-19:
I also want to know.

muzi-15, 2015-05-15:
till now the PCA9685 prototype can work

onceme, 2015-09-01:
@muzi-15 consider to close this issue

droneboost, 2015-09-07:
Since BH first edition has come out, I close this issue.

closed 2015-09-07 by droneboost