bithollow/bitforge #16: [M1][High][HW]  ALL in one  prototype(sensor hub)

muzi-15, 2015-05-15:


muzi-15, 2015-05-15:
Sensor:
  
    MPU6050  IMU
    MS5611 Barometer
    U-blox M8N Glonass/GPS/Beidou
    ADS1115 ADC for power monitoring
    MB85RC FRAM storage
    HAT EEPROM
    PCA9685 PWM generator
    RGB LED

muzi-15, 2015-05-15:
PWM servo outputs
    PPM input
    PWM input
    Triple redundant power supply
    Power module connector
    UART, SPI, I2C for extensions
    HAT compliant

muzi-15, 2015-05-26:
GPS SELECT:
1. http://u-blox.com/en/gps-modules/pvt-modules/neo-7.html
2. http://www.u-blox.com/en/gps-modules/pvt-modules/previous-generations/neo-6-family.html
3. http://www.u-blox.com/images/downloads/Product_Docs/GNSS-module-selector_Overview_%28UBX-14000426%29.pdf

muzi-15, 2015-05-27:
1. RTC
2. LED
3. Voltage detect
4. Buzzer
5. 5V/3/3V DC converter
6. Moderate temperature detect

onceme, 2015-06-03:
HMC5883L

onceme, 2015-09-01:
Now we have breadboard and BH board, consider this one closed?

droneboost, 2015-09-07:
Since BH first edition has come out, I close this issue.

closed 2015-09-07 by droneboost