bithollow/bitforge #19: [M?][Medium][APM] Start APM automatically after RPI2 boot

onceme, 2015-05-30:
Start APM automatically after RPI2 boot instead of starting it through serial console or ssh shell.

- Study how to add systemd service
- Add APM service
- Consider how to connect -A telemetry, currently we are using static IP addresses

onceme, 2015-06-07:
refer to this git, will be summarized to wiki later.
https://gist.github.com/onceme/0575eb651e8d28301bf9

onceme, 2015-06-08:
found in field test that if the specified ip address cannot be reached, ArduCopter will not get ESC ready (the 'ready' beep sound won't fire).

My RPI2 usb wifi was set as 'dhcp', but while field testing, there was no hotspot to lease IP to RPI2. So I'm not sure whether it was because RPI2 would wait until some IP address is acquired, then start ArduCopter, or, ArduCopter was started, but it was waiting for the unreachable IP address to respond.

droneboost, 2015-09-07:
@onceme , close this issue?

onceme, 2015-09-07:
auto starting APM has been integrated to rpi.img.bz2

closed 2015-09-07 by onceme