bithollow/bitforge #20: [M?][Medium][APM] Enable APM -C telemetry

onceme, 2015-05-30:
APM should be able to connect GCS through -C telemetry, which is over serial port, but in RPI2, there is only one serial port, it is used as RPI2 serial console. Find a way to let APM use /dev/ttyAMA0 in RPI2 as -C telemetry.

- Modify cmdline.txt to disable RPI2 serial console
- Build APM with serial port /dev/ttyAMA0
- Verify it works

onceme, 2015-06-07:
Places needs to be changed:

- $ARDUPILOT_SRC_ROOT/config.mk, "PORT = /dev/ttyAMA0"
- In RPI rootfs, /boot/firmware/cmdline.txt, remove "console=ttyAMA0,115200 kgdboc=ttyAMA0,115200"

When startup ArduCopter.elf, try the following:
```shell
$ sudo ./ArduCopter.elf -A /dev/ttyAMA0
```shell

Yes, probably you have already noticed, here we use telemetry -A instead of -C, for now -C will not work, reason unknown.

onceme, 2015-06-07:
Found this can work:
$ sudo ./ArduCopter.elf -A /dev/ttyAMA0 -C udp:xxx.xxx.xxx.xxx:14550

By connecting serial cable and Ethernet at the same time, they work simultaneously. Either unplugging serial cable or disable Ethernet, the other one can continuously work.

-A telemetry is the default telemetry, I guess serial port should come with default. But from the actually code, I didn't find out such limitation, -A telemetry and -C telemetry could be specified as user want. Big question mark here.

droneboost, 2015-09-07:
@onceme , close this issue?

onceme, 2015-09-07:
Now all telemetries are enabled

closed 2015-09-07 by onceme