bithollow/bitforge #24: [M?][High][HW] Collect retailer/vendor information for BOM

onceme, 2015-08-04:
Collect parts information before we actually purchase them.
- Main parts (onceme)
- Connectors (?)
- Miscellaneous (?) 

Add retail link here.

onceme, 2015-08-05:
# Main parts

## 纳达尔数码专营店
### PCA9685BS
http://detail.tmall.com/item.htm?spm=a230r.1.14.42.CSOT08&id=39448841343&ns=1&abbucket=19

## 深圳市江东电子商行
### NEO-M8N-0-01
http://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-11259359514.11.UFgpft&id=42554700868
### MS5611-01BA
http://item.taobao.com/item.htm?spm=2013.1.w4018-11259379161.7.bC4LoF&id=14157565240&scm=1007.11837.6006.i39903117808&pvid=45149f6c-7721-4726-baab-ed235cbcbf6e
### MPU-6050C
http://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-11259359514.23.4fWn2Q&id=35895053463
### HMC5883L-TR
http://item.taobao.com/item.htm?spm=a230r.1.14.19.NRCfpE&id=39903117808&ns=1&abbucket=19#detail
### MS621FE
http://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-11259359514.9.JEwwE5&id=17111394953

## 金利达微 少女show
### DS1339U-3
http://world.taobao.com/item/45065635391.htm?spm=a312a.7700714.0.0.nqDQco#detail
### SP6205EM5
http://item.taobao.com/item.htm?spm=a312a.7700824.w4002-1070494141.9.SOoWki&id=15423878212

## 深圳鑫达电子
### WS2812B
http://item.taobao.com/item.htm?spm=a230r.1.14.73.1mx27W&id=45488927580&ns=1&abbucket=19#detail

muzi-15, 2015-08-11:
Diode	3.5X2.8X1.9	1	D3( IN5817  or IN4148)

muzi-15, 2015-08-11:
PCA9685BS 4?*10=40
ADS1115IDGSR/T 4?*13=52
NEO-M8N-0-01 4?*98=392
MS5611-01BA 4? *28=112
MPU-6050C 4?*15=60
HMC5883L-TR 4?*5=20
DS1339U-3 4?  *6=20
MS621FE MS621 4? *3=12

droneboost, 2015-08-11:
Date| Goods | Quantity  | Components | Expenditure  | Payer
------- | ------- | -------- | ---------- | -------- | ------
8/3 | BH board 1st edition + Transport fee | 10 | 193 | Onceme, Muzu-15, D.B.

muzi-15, 2015-08-12:
2.54mm 3×40pin的， 10只 X 3 = 30元
2.54mm  2×20pin 贴片排母  5只 x1.5 = 7.5
2.54  2x20pin 贴片排针  5只 X1.5 = 7.5
共阳 RGB 3528发光二极管  20 只 =5元
1K 电阻 0603 30只  100个2元
0.1uf 电容 0603封装 100只  100个3元
0.01uf 电容 0603封装 30只  100个3元
10uf 电容 0603封装 30只  100个5元
27nh 电感 0603封装 30只   100个4元
4R7 电感  5只5元 
1.25mm间距 卧贴  4P 24 6P 12

muzi-15, 2015-08-12:
All the elements is on their way to BH board.

droneboost, 2015-08-25:
# Main parts

## 纳达尔数码专营店
### [PCA9685BS](http://detail.tmall.com/item.htm?spm=a230r.1.14.42.CSOT08&id=39448841343&ns=1&abbucket=19)
### [ADS1115IDGSR/T](http://world.tmall.com/item/38834846102.htm?spm=a312a.7700714.0.0.wDcbjl)

## 单片机snq2010
### [ADS1115IDGSR/T](http://world.taobao.com/item/13163619405.htm?spm=a312a.7700714.0.0.wDcbjl#detail)

## 深圳市江东电子商行
### [NEO-M8N-0-01](http://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-11259359514.11.UFgpft&id=42554700868)
### [MS5611-01BA](http://item.taobao.com/item.htm?spm=2013.1.w4018-11259379161.7.bC4LoF&id=14157565240&scm=1007.11837.6006.i39903117808&pvid=45149f6c-7721-4726-baab-ed235cbcbf6e)
### [MPU-6050C](http://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-11259359514.23.4fWn2Q&id=35895053463)
### [HMC5883L-TR](http://item.taobao.com/item.htm?spm=a230r.1.14.19.NRCfpE&id=39903117808&ns=1&abbucket=19#detail)
### [MS621FE](http://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-11259359514.9.JEwwE5&id=17111394953)

## 金利达微 少女show
### [DS1339U-3](http://world.taobao.com/item/45065635391.htm?spm=a312a.7700714.0.0.nqDQco#detail)
### [SP6205EM5](http://item.taobao.com/item.htm?spm=a312a.7700824.w4002-1070494141.9.SOoWki&id=15423878212)

## 深圳鑫达电子
### [WS2812B](http://item.taobao.com/item.htm?spm=a230r.1.14.73.1mx27W&id=45488927580&ns=1&abbucket=19#detail)

# Connectors 

# Miscellaneous

droneboost, 2015-08-25:
Diode 3.5X2.8X1.9 1 D3( IN5817 or IN4148)

droneboost, 2015-08-25:
PCA9685BS 4?10=40
ADS1115IDGSR/T 4?13=52
NEO-M8N-0-01 4?98=392
MS5611-01BA 4? *28=112
MPU-6050C 4?15=60
HMC5883L-TR 4?*5=20
DS1339U-3 4? *6=20
MS621FE MS621 4? *3=12

droneboost, 2015-09-07:
@onceme, If this page is only for BH first edition, I'd like to close this issue.

onceme, 2015-09-07:
Yes, it's for BH first edition, I will close it.

closed 2015-09-07 by onceme