bithollow/bitforge #25: [M?][High][SW] Verify APM 3.3 merged by Aaron

droneboost, 2015-08-25:
D.B. will verify it.

onceme, 2015-09-01:
How is this going? I have updated Copter-3.3-BH-develop, please update your local copy

onceme, 2015-09-06:
The crash has been fixed, commit is here: https://github.com/bithollow/ardupilot/commit/86f97f31ab8c91f089664bf1b2ce36a0af2138b1

Please continue to check other parts. I don't have the board in handle, from GCS, the MPU6050 data looks strange.

onceme, 2015-09-06:
The crash was caused by other's change, not ours. It is in Compass.cpp

droneboost, 2015-09-07:
I've tested the latest Copter-3.3-BH-develop code without ESC mounted, it seems work well,GCS status tracing is very stable, but we still need to test it with ESC and motor before switch this branch to main dev branch.

onceme, 2015-09-28:
Please test this based on kernel 4.1.5/APM Copter-3.3 rc12, don't wait for me, I don't know when they would update kernel/APM again, so till the due date of M1, 4.1.5/rc12 might be the final software version.

tick tock, tick tock...

droneboost, 2015-09-29:
Sensor data is not stable on kernel 4.1.5(rt) + latest Debian jessie rootfs.

droneboost, 2015-10-31:
estimate 3 days to finish the test flight with Muzu-15.

droneboost, 2015-12-03:
Copter-3.3-BH-Develop:
D.B.'s 0.2 board: restart ardupilot.service several times before ardupilot work normally.
Muzu-15's 0.2 board: ardupilot.service keeps restarting, never work normally.

Master:
D.B. and Muzu-15's 0.2 board both work normally.

RC calibration in processing...

droneboost, 2015-12-25:
Test flight has been done and movie has been recored.
APM seems very stable in spite of operator's suck skill.
I close this issue.

closed 2015-12-25 by droneboost

onceme, 2015-12-25:
Your first flight looks better than mine:)