bithollow/bitforge #26: [M?][SW] Improve debian bootup time and tailor kernel

droneboost, 2015-08-31:


droneboost, 2015-09-16:
Deadline is 10/30

droneboost, 2015-10-31:
Estimate 3 days to finish this issue.

droneboost, 2015-11-04:
Just memo, will make a reference to the following page later:
https://wiki.archlinux.org/index.php/Improve_boot_performance

droneboost, 2016-01-28:
Since this is a trivial task in M2 and M2 is delayed for a long time, I suggest to defer this issue to next Milestone, how do you think, BH guys?

onceme, 2016-01-28:
I suggest remove this issue from Milestone for now. Once we have time to fine tune BH sw, then pull it back.

muzi-15, 2016-01-28:
OK， got it. defer to next Mil..

muzi-15, 2016-12-06:
need about 10 days.

muzi-15, 2017-03-28:
ETA:4/165

muzi-15, 2017-05-24:
eth0: dhcp 
boot time:
      1min 6.893s networking.service
         10.117s rsyslog.service
          1.467s dnsmasq.service
          1.131s systemd-logind.service
           723ms lighttpd.service
           706ms resolvconf.service
           703ms systemd-udev-trigger.service
           630ms hostapd.service
           513ms rc-local.service
           507ms systemd-setup-dgram-qlen.service
           468ms dhcpcd.service
           434ms systemd-modules-load.service
           324ms kmod-static-nodes.service
           314ms systemd-journal-flush.service
           313ms systemd-random-seed.service
           298ms systemd-tmpfiles-setup-dev.service
           287ms systemd-remount-fs.service
           248ms systemd-sysctl.service
           241ms dev-mqueue.mount
           193ms sys-kernel-config.mount
           176ms systemd-user-sessions.service
           168ms sys-kernel-debug.mount
           168ms hciuart.service
           142ms systemd-tmpfiles-setup.service
           102ms bluetooth.service
            95ms hwclockfirst.service
            70ms systemd-update-utmp.service
            67ms udev-finish.service
            63ms systemd-update-utmp-runlevel.service
            47ms boot-firmware.mount
            41ms systemd-rfkill@rfkill0.service
            37ms systemd-udevd.service
            23ms systemd-rfkill@rfkill1.service

muzi-15, 2017-05-24:
eth0: manual
 boot time:
Startup finished in 3.427s (kernel) + 16.768s (userspace) = 20.196s
detailed time:
          1.684s dnsmasq.service
           961ms lighttpd.service
           950ms systemd-logind.service
           847ms dhcpcd.service
           847ms networking.service
           787ms resolvconf.service
           785ms systemd-udev-trigger.service
           667ms hostapd.service
           577ms systemd-setup-dgram-qlen.service
           528ms rc-local.service
           496ms systemd-modules-load.service
           458ms hwclockfirst.service
           330ms kmod-static-nodes.service
           313ms systemd-random-seed.service
           300ms dev-mqueue.mount
           298ms systemd-tmpfiles-setup-dev.service
           297ms systemd-remount-fs.service
           228ms sys-kernel-debug.mount
           213ms systemd-sysctl.service
           194ms sys-kernel-config.mount
           176ms hciuart.service
           172ms rsyslog.service
           151ms boot-firmware.mount
           124ms systemd-journal-flush.service
           102ms bluetooth.service
            70ms systemd-user-sessions.service
            53ms systemd-update-utmp-runlevel.service
            52ms systemd-update-utmp.service
            49ms systemd-tmpfiles-setup.service
            47ms udev-finish.service
            37ms systemd-udevd.service
            33ms systemd-rfkill@rfkill0.service
            21ms systemd-rfkill@rfkill1.service

muzi-15, 2017-05-24:
most of the start time has spent on networking.service starting. about 1minute.

muzi-15, 2017-10-19:
current bootup time was acceptable, so there is no need keep on improve it.

closed 2017-10-19 by muzi-15