bithollow/bitforge #27: [M2][High][HW] Design BH box cover

droneboost, 2015-08-31:


onceme, 2015-10-31:
Please add Background/TODOs/Outputs

droneboost, 2015-11-04:
New finding!

Autodesk released a free and easy using 3D modeling tools - Autodesk 123D. 
With the free policy, people are forging an 3D design community, where people can offering free or no-free 3D modeling designs, for raspberrypi, I found a free 3D modeling design at http://www.123dapp.com/123D_Design/Pi-case/3218365, seems we can re-use this model to customize BH protect case.

droneboost, 2015-11-04:
googled 3D print price, priced in per gram unit.

[image]

onceme, 2015-11-04:
There a lot of 3d printers on hobbyking with good price

onceme, 2015-11-04:
Your googled price is for the material, not the printer itself.

droneboost, 2015-11-04:
3. Find a 3D printing service provider on TaoBao or Tmall to print the case.

muzi-15, 2016-12-06:
it is hard to evaluate details workdays, generally 20 days

muzi-15, 2017-02-23:
The Price was bellow:

huangzetao888 (13:20:07):
这个大量加工的话一个大概是40元左右
单独做几个的话出样费大概400元左右
yongxinwu_2008 (13:23:54):
了解，多谢

muzi-15, 2017-03-28:
ETA: 4/15

muzi-15, 2017-05-22:
the box design has finished, there are total 4 pieces.

droneboost, 2017-05-22:
Good!

droneboost, 2017-05-22:
Show us the picture

muzi-15, 2017-05-23:
it seems can not upload pictures. please see it by  wechat.

muzi-15, 2017-05-23:
[image]

droneboost, 2017-05-23:
check comments in your input textbox

 "Attach files by dragging & dropping,  选择文件 selecting them, or pasting from the clipboard."

droneboost, 2017-05-23:
[image]

muzi-15, 2017-05-24:
[image]

closed 2017-05-24 by muzi-15

droneboost, 2017-05-24:
Good!

droneboost, 2017-05-31:
Have you upload the design file to bitbucket?