bithollow/bitforge #28: [M2] BH board beta ersion‘ delivery

muzi-15, 2015-09-01:
1.  2 board: A for RPI2 and B for MCU(like stm32).
2.  6050 or 9250
3.  drivers packages and demos for BH board

onceme, 2015-09-01:
> 3. drivers packages and demos for BH board
I think this could be put in another issue, and more details are required. Such as driver packages list, demo list etc.

muzi-15, 2015-09-01:
OK, it is good, it should be split into some features and more issues

onceme, 2015-09-01:
2 boards: A for RPI2 and B for MCU(like stm32).

model A:
definition: it is the updated and fine tuned version of current BH, could be delivered to people other than the three of us
- a decent board name (I think "BH" is just temporary)
- use MPU9250 SPI
- put single color leds closer to each other
- black or blue PCB
- 40 long pins male connector to 2 of 40 pins female connectors (on the top and bottom separately)
- remove jump wires
- change D4 to red, D5 to yellow, D6 to blue, D1 to white
- connect D1 to GPS directly 
- change crystal oscillator to SMT version, support devices which can use external crystal oscillator

model B:
definition: it is the MCU equipped version of model A, the MCU can be used as standalone pilot software processor
- add MCU (stm32?)

onceme, 2015-10-31:
Three owners all agree to close, because module B canceled

closed 2015-10-31 by onceme