bithollow/bitforge #30: [M1][SW] Integrate wifi + avahi daemon + RTC control control to rootfs

droneboost, 2015-09-16:
Include:

SoftAP
Lighttpd
Avahi
RTC

droneboost, 2015-09-24:
Update my work status:
1. Build latest rt kernel 4.1.5   <= Done 9/24
2. Create latest clean Debian Jessie rootfs   <= Done 9/24
3. Integrate avahi/lighttpd <= In processing 9/24
4. Integrate wpa_supplicant
5. Integrate hostapd
6. Integrate RTC
7. Other stuffs

onceme, 2015-10-22:
this is my local image, tried 3 times, almost the same, booted and connected around 10 seconds
```shell
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.9-rt8-v7+ (bit@aws-autopilot) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1 SMP PREEMPT RT Thu Oct 8 08:42:44 UTC 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: BCM2709
[    0.000000] cma: Reserved 8 MiB at 0x3d800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (9460->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @bcefc000 s21824 r8192 d23232 u53248
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa21041 bcm2709.serial=0xd2ecb3e1 smsc95xx.macat
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 987504K/1015808K available (6144K kernel code, 535K rwdata, 1672K rodata, 428K init, 780K bss, 20112K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff000000   (1032 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe000000   ( 992 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x807aa280   (7817 kB)
[    0.000000]       .init : 0x807ab000 - 0x80816000   ( 428 kB)
[    0.000000]       .data : 0x80816000 - 0x8089bd70   ( 536 kB)
[    0.000000]        .bss : 0x8089e000 - 0x80961100   ( 781 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  Additional per-CPU info printed with stalls.
[    0.000000]  RCU kthread priority: 1.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000012] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000322] Console: colour dummy device 80x30
[    0.000605] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000613] pid_max: default: 32768 minimum: 301
[    0.001022] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001038] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002512] Initializing cgroup subsys blkio
[    0.002535] Initializing cgroup subsys memory
[    0.002568] Initializing cgroup subsys devices
[    0.002589] Initializing cgroup subsys freezer
[    0.002629] Initializing cgroup subsys net_cls
[    0.002795] CPU: Testing write buffer coherency: ok
[    0.002808] ftrace: allocating 20267 entries in 60 pages
[    0.057360] missing device node for CPU 0
[    0.057454] missing device node for CPU 1
[    0.057469] missing device node for CPU 2
[    0.057483] missing device node for CPU 3
[    0.057507] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.057526] [bcm2709_smp_prepare_cpus] enter
[    0.057659] Setting up static identity map for 0x8240 - 0x8274
[    0.177459] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.177894] [bcm2709_secondary_init] enter cpu:1
[    0.177953] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.217464] [bcm2709_boot_secondary] cpu:2 started (0) 18
[    0.217816] [bcm2709_secondary_init] enter cpu:2
[    0.217852] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.257551] [bcm2709_boot_secondary] cpu:3 started (0) 17
[    0.257807] [bcm2709_secondary_init] enter cpu:3
[    0.257839] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.257950] Brought up 4 CPUs
[... 343 lines cut ...]
bithollow login: [   17.198756] cfg80211: Calling CRDA to update world regulatory domain
[   20.358753] cfg80211: Calling CRDA to update world regulatory domain
[   23.518733] cfg80211: Calling CRDA to update world regulatory domain
[   26.678738] cfg80211: Calling CRDA to update world regulatory domain
[   29.838763] cfg80211: Calling CRDA to update world regulatory domain
[   32.998743] cfg80211: Calling CRDA to update world regulatory domain
[   36.158771] cfg80211: Calling CRDA to update world regulatory domain
[   39.318755] cfg80211: Calling CRDA to update world regulatory domain
[   42.478774] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA

Debian GNU/Linux 8 bithollow ttyAMA0

bithollow login: bit
Password: 
Last login: Thu Jan  1 00:00:57 UTC 1970 on ttyAMA0
Linux bithollow 4.1.9-rt8-v7+ #1 SMP PREEMPT RT Thu Oct 8 08:42:44 UTC 2015 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
bit@bithollow:~$ 
```

onceme, 2015-10-22:
I found several problems in your image
- /etc/wpa_supplicant.conf has incorrect wifi ssid and password
- /etc/network/interfaces, eth0 is set to 'dhcp', if network cable is not connected, it will take about 2 minutes to timeout for eth0, then start wlan0 dhcp
- shell response is much more slow than my local image
- there are DHCP error

```shell
[FAILED] Failed to start LSB: DHCP server.
See 'systemctl status isc-dhcp-server.service' for details.
```

After I set my own wifi ssid and pass, installed 8188eu driver, reboot, I can connect to my wifi router in about 26 seconds. The full log:

```shell
bit@bithollow:~$ 
bit@bithollow:~$ ls
8188eu.conf  ardupilot  bh_daemon    install.sh     pigpio
8188eu.ko    aws_temp   hostapd_src  lighttpd_fcgi  wifi_driver
bit@bithollow:~$ sudo ./install.sh 
sudo: unable to resolve host bithollow
sudo cp 8188eu.conf /etc/modprobe.d/.
sudo: unable to resolve host bithollow
sudo install -p -m 644 8188eu.ko /lib/modules/4.1.9-rt8-v7+/kernel/drivers/net/wireless
sudo: unable to resolve host bithollow
sudo depmod 4.1.9-rt8-v7+
sudo: unable to resolve host bithollow

Reboot to run the driver.

If you have already configured your wifi it should start up and connect to your
wireless network.

If you have not configured your wifi you will need to do that to enable the wifi.
bit@bithollow:~$ sudo reboot
sudo: unable to resolve host bithollow
         Starting Synchronise Hardware Clock to System Clock...
[  OK  ] Stopped target Network is Online.
[  OK  ] Stopped target Graphical Interface.
[  OK  ] Stopped target Multi-User System.
         Stopping Lighttpd Daemon...
         Stopping ArduPilot on BH...
         Stopping Bithollow Daemon Service...
         Stopping Regular background program processing daemon...
         Stopping OpenBSD Secure Shell server...
[  OK  ] Stopped target Login Prompts.
         Stopping Getty on tty1...
         Stopping Serial Getty on ttyAMA0...
         Stopping Login Service...
         Stopping Avahi mDNS/DNS-SD Stack...
         Stopping D-Bus System Message Bus...
         Stopping System Logging Service...
[  OK  ] Stopped Bithollow Daemon Service.
[  OK  ] Stopped Regular background program processing daemon.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Avahi mDNS/DNS-SD Stack.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Getty on tty1.
[  OK  ] Stopped Serial Getty on ttyAMA0.
[  OK  ] Stopped Lighttpd Daemon.
[  OK  ] Stopped OpenBSD Secure Shell server.
[  OK  ] Stopped ArduPilot on BH.
[  OK  ] Removed slice system-serial\x2dgetty.slice.
[  OK  ] Removed slice system-getty.slice.
         Stopping /etc/rc.local Compatibility...
[  OK  ] Stopped /etc/rc.local Compatibility.
[  OK  ] Stopped target Network.
         Stopping Permit User Sessions...
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped target Basic System.
[... 1065 lines cut ...]
bit@bithollow:~$ 
bit@bithollow:~$ 
bit@bithollow:~$ sudo ifconfig
[sudo] password for bit: 
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:12 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:704 (704.0 B)  TX bytes:704 (704.0 B)

wlan0     Link encap:Ethernet  HWaddr 3c:46:d8:1e:d5:95  
          inet addr:192.168.0.102  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::3e46:d8ff:fe1e:d595/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:49 errors:0 dropped:16 overruns:0 frame:0
          TX packets:75 errors:0 dropped:27 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7627 (7.4 KiB)  TX bytes:9674 (9.4 KiB)

bit@bithollow:~$ 
 CTRL-A Z for help |115200 8N1 | NOR | Minicom 2.4    | VT102 |      Offline                                                                                            
```

droneboost, 2015-10-25:
bh0.2_linux4.1.9-rt8-v7+.RC01.bz2 has been pushed to bithollow/bitburger/sdimages/rpi2/bh0.2_linux4.1.9-rt8-v7+.RC01.bz2

In this edition, still need to manually install 8188eu.ko after first time boot up.

onceme, 2015-10-26:
memo about bh0.2_linux4.1.9-rt8-v7+.RC01.bz2

generally speaking, I think this image is for users, it should be 'automatic' and 'minimum'.

problems I met:
- there are source code and development packages inside, should be removed
- bh_daemon should be changed as: GPIO 24 / 3.3v shorted: station mode; non-shorted: softap mode
- user needs to install 8188eu driver manually after first boot
- when in softap mode, iphone6/sony xperia/ubuntu 10.4 pc can't connect to bithollow, they notice me "password is incorrect"; SONY laptop with win7 can connect, but can't acquire IP from DHCP, dhcp server start failed in boot message. Because of this, I can't try softap with GCS
- need wiki page or gist to teach how to short GPIO 24 / 3.3v
- I think avahi is not necessary this time, uart is enough for now if the 'user' is a developer. Connect network cable and serial cable consume similar efforts. So we can keep it 'minimum'. More, user has wifi.

droneboost, 2015-10-26:
Just memo:

To avoid "cfg80211: Calling CRDA to update world regulatory domain" issue on bootup,

sudo apt-get install wireless-regdb iw crda

Reason:
Central Regulatory Domain Agent (CRDA) is used by the Linux kernel cfg80211 wireless subsystem to query and apply the regulatory domain settings wireless devices may operate within for a given location. Not sure why Jessie distro does not have these included, but Wheezy distro.

Reference:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=122067
https://packages.debian.org/jessie/crda

droneboost, 2015-10-26:
Just memo, 8818eu.ko crash again in softap mode:

root@bithollow:/home/bit#
Entering kdb (current=0xbcbb57c0, pid 65) on processor 0 Oops: (null)
due to oops @ 0x80589f18
CPU: 0 PID: 65 Comm: irq/75-dwc_otg_ Tainted: G           O    4.1.9-rt8-v7+ #1
Hardware name: BCM2709
task: bcbb57c0 ti: bc08c000 task.ti: bc08c000
PC is at rt_spin_lock_slowlock+0x2d4/0x310
LR is at 0xbf04b4f4
pc : [<80589f18>]    lr : [<bf04b4f4>]    psr: 600b0013
sp : bc08dbd8  ip : bcbb57c1  fp : bc08dc2c
r10: bc17a800  r9 : bf07601c  r8 : 00000000
r7 : bcbb57c0  r6 : bc08dbd8  r5 : 00000001  r4 : bf04b4e8
r3 : 00000001  r2 : bcbb57c0  r1 : bcbb57c0  r0 : 00000000
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Control: 10c5387d  Table: 3a15806a  DAC: 00000015
CPU: 0 PID: 65 Comm: irq/75-dwc_otg_ Tainted: G           O    4.1.9-rt8-v7+ #1
Hardware name: BCM2709
[<800188e0>] (unwind_backtrace) from [<80014194>] (show_stack+0x20/0x24)
[<80014194>] (show_stack) from [<80587394>] (dump_stack+0x88/0x9c)
[<80587394>] (dump_stack) from [<80010f08>] (show_regs+0x1c/0x20)
[<80010f08>] (show_regs) from [<800bd4e8>] (kdb_main_loop+0x3ec/0x800)
[<800bd4e8>] (kdb_main_loop) from [<800c03d0>] (kdb_stub+0x2c4/0x510)
[<800c03d0>] (kdb_stub) from [<800b5ce8>] (kgdb_cpu_enter+0x480/0x8f0)
[<800b5ce8>] (kgdb_cpu_enter) from [<800b63b4>] (kgdb_handle_exception+0x104/0x218)
[<800b63b4>] (kgdb_handle_exception) from [<80017de4>] (kgdb_notify+0x40/0x90)
[<80017de4>] (kgdb_notify) from [<80045650>] (notifier_call_chain+0x54/0x94)
[<80045650>] (notifier_call_chain) from [<8004599c>] (atomic_notifier_call_chain+0x44/0x58)
[<8004599c>] (atomic_notifier_call_chain) from [<800459f4>] (notify_die+0x44/0x4c)
[<800459f4>] (notify_die) from [<800142a4>] (die+0x10c/0x430)
[<800142a4>] (die) from [<800145f4>] (arm_notify_die+0x2c/0x64)
[<800145f4>] (arm_notify_die) from [<80009150>] (do_undefinstr+0x150/0x1a0)
[<80009150>] (do_undefinstr) from [<8058bff0>] (__und_svc_finish+0x0/0x30)
Exception stack(0xbc08db50 to 0xbc08db98)
db40:                                     00000000 bcbb57c0 bcbb57c0 00000001
db60: bf04b4e8 00000001 bc08dbd8 bcbb57c0 00000000 bf07601c bc17a800 bc08dc2c
db80: bcbb57c1 bc08dbd8 bf04b4f4 80589f18 600b0013 ffffffff
[<8058bff0>] (__und_svc_finish) from [<80589f18>] (rt_spin_lock_slowlock+0x2d4/0x310)
[<80589f18>] (rt_spin_lock_slowlock) from [<8058b790>] (rt_spin_lock+0x58/0x5c)
[<8058b790>] (rt_spin_lock) from [<7f3bbb44>] (usb_write_port+0x100/0x350 [8188eu])
[<7f3bbb44>] (usb_write_port [8188eu]) from [<7f34e684>] (_rtw_write_port+0x28/0x2c [8188eu])
[<7f34e684>] (_rtw_write_port [8188eu]) from [<7f3a0c0c>] (rtw_dump_xframe+0x1e4/0x290 [8188eu])
[<7f3a0c0c>] (rtw_dump_xframe [8188eu]) from [<7f3a12dc>] (rtl8188eu_mgnt_xmit+0x18/0x1c [8188eu])
[<7f3a12dc>] (rtl8188eu_mgnt_xmit [8188eu]) from [<7f38aa74>] (rtw_hal_mgnt_xmit+0x28/0x34 [8188eu])
[<7f38aa74>] (rtw_hal_mgnt_xmit [8188eu]) from [<7f35a944>] (dump_mgntframe+0x38/0x3c [8188eu])
[<7f35a944>] (dump_mgntframe [8188eu]) from [<7f3608b4>] (issue_nulldata+0x1ac/0x32c [8188eu])
[<7f3608b4>] (issue_nulldata [8188eu]) from [<7f372568>] (rtw_recv_entry+0x84c/0x1894 [8188eu])
[<7f372568>] (rtw_recv_entry [8188eu]) from [<7f3a5ea8>] (rtl8188eu_recv_tasklet+0x4b8/0x69c [8188eu])
[<7f3a5ea8>] (rtl8188eu_recv_tasklet [8188eu]) from [<8002b6c8>] (__tasklet_action.isra.19+0x70/0x134)
[<8002b6c8>] (__tasklet_action.isra.19) from [<8002b8b0>] (tasklet_action+0x90/0x94)
[<8002b8b0>] (tasklet_action) from [<8002b190>] (do_current_softirqs+0x210/0x3a8)
[<8002b190>] (do_current_softirqs) from [<8002b3c0>] (__local_bh_enable+0x98/0xb0)
[<8002b3c0>] (__local_bh_enable) from [<8002b3f0>] (local_bh_enable+0x18/0x1c)
[<8002b3f0>] (local_bh_enable) from [<8007479c>] (irq_forced_thread_fn+0x50/0x64)
[<8007479c>] (irq_forced_thread_fn) from [<80074a9c>] (irq_thread+0x164/0x23c)
[<80074a9c>] (irq_thread) from [<80044680>] (kthread+0xdc/0xf4)
[<80044680>] (kthread) from [<8000fcf8>] (ret_from_fork+0x14/0x3c)

droneboost, 2015-11-02:
Update:

Tried,
1. remove source code,
2. apt-get autoremove --purge git build-essential ncurses-dev bc file tree
3. apt-get clean

The rootfs image grows to 404mb...Trying to create a clean and minimal rootfs structure now.

droneboost, 2015-11-04:
rootfs size before write into rpi.img, will check real size on board.

```shell
# du -shx *
5.4M    bin
0       boot
0       dev
2.3M    etc
145M    home  #contains source and git repo, will be removed when mksd6.sh
28M     lib
0       media
0       mnt
0       opt
0       proc
16K     root
8.0K    run
4.7M    sbin
0       srv
0       sys
0       tmp
363M    usr
155M    var
```

onceme, 2015-11-10:
When can you actually close this issue?

droneboost, 2015-11-17:
Upload bh2.0-linux4.1.9-v7-rt8-4.1.7.img.bz2 to bitburger git repo.

For internal use, please retrieve this image file from aws@43.82.111.228:~/Works/Fei.Cao/cross-compile-rootfs/home/bit/bitburger/sdimages/rpi2/bh2.0-linux4.1.9-v7-rt8-4.1.7.img.bz2

If any question, please contact D.B.

closed 2015-11-17 by droneboost