bithollow/bitforge #34: [M1][High] TL-WN725N usb wifi dongle in Linux 4.1.x

onceme, 2015-09-16:
- Background
Currently the usb wifi dongle driver is manually added into rootfs, and it is for kernel 3.18.9. After update to more higher version, this driver have to be integrated manually again each time.

- TODOs
  - prepare the wifi driver for new kernel
  - make the wifi-integration process as automatic as possible

- Output
  - wifi driver binaries
  - auto script

onceme, 2015-10-10:
@droneboost @muzi-15

- download this package into RPI2 at runtime
- extract the package
- execute install.sh ($ sudo path/to/install.sh)

If you want to integrate it into rpi.img.bz2, install it in RPI2 first, and copy the changed files to rootfs

droneboost, 2015-10-11:
Jan  1 00:00:18 bithollow systemd-modules-load[125]: Failed to insert '8188eu': Exec format error

onceme, 2015-10-11:
@droneboost Install.sh is a must

droneboost, 2015-10-12:
What I have done:

On RPI,

```shell
su root
cd ~/tools/pkg
tar jvxf 8188eu-rpi-4.1.9-v7-preempt-rt8-4.1.7.tar.bz2
./install.sh

ls -al /lib/modules/4.1.9-rt8-v7+/kernel/drivers/net/wireless/
total 928
drwxr-xr-x 16 root root   4096 Oct 11 15:44 .
drwxr-xr-x 13 root root   4096 Oct  5 03:15 ..
-rw-r--r--  1 root root 671336 Oct 10 10:02 8188eu.ko
-rw-r--r--  1 root root  59696 Oct  5 03:15 at76c50x-usb.ko
drwxr-xr-x  6 root root   4096 Oct  5 03:15 ath
drwxr-xr-x  2 root root   4096 Oct  5 03:15 b43
drwxr-xr-x  2 root root   4096 Oct  5 03:15 b43legacy
drwxr-xr-x  4 root root   4096 Oct  5 03:15 brcm80211
drwxr-xr-x  2 root root   4096 Oct  5 03:15 hostap
drwxr-xr-x  2 root root   4096 Oct  5 03:15 libertas
drwxr-xr-x  2 root root   4096 Oct  5 03:15 libertas_tf
-rw-r--r--  1 root root  67056 Oct  5 03:15 mac80211_hwsim.ko
drwxr-xr-x  3 root root   4096 Oct  5 03:15 mediatek
drwxr-xr-x  2 root root   4096 Oct  5 03:15 mwifiex
drwxr-xr-x  2 root root   4096 Oct  5 03:15 p54
-rw-r--r--  1 root root  47140 Oct  5 03:15 rndis_wlan.ko
drwxr-xr-x  2 root root   4096 Oct  5 03:15 rt2x00
drwxr-xr-x  3 root root   4096 Oct  5 03:15 rtl818x
drwxr-xr-x  2 root root   4096 Oct  5 03:15 rtl8192cu
-rw-r--r--  1 root root  31320 Oct  5 03:16 zd1201.ko
drwxr-xr-x  2 root root   4096 Oct  5 03:16 zd1211rw
```

onceme, 2015-10-12:
The problem is the stripped kernel model could not be used. I packed non-stripped version again in this commit: acc5b272e13af7041023e71ad9d8b5ad7b474ed5

droneboost, 2015-10-13:
Great job! It works now! 

```shell
root@bithollow:/home/bit# lsmod
Module                  Size  Used by
cfg80211              415263  0
rfkill                 15829  2 cfg80211
8188eu                738660  0
............

root@bithollow:/home/bit# ifconfig
wlan0     Link encap:Ethernet  HWaddr 3c:46:d8:1b:cb:b6
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

closed 2015-10-15 by onceme