bithollow/bitforge #40: [M2][Low] Merge our changes to ardupilot.git to upstream

droneboost, 2015-10-08:


onceme, 2015-10-31:
I need at least 15 days to finish this if APM maintainer agreed.

onceme, 2015-11-07:
APM upstream developers agreed to accept our pull request.

https://groups.google.com/forum/#!topic/drones-discuss/oE1alXaXNzM

onceme, 2015-11-07:
latest master branch is prepared

onceme, 2015-11-08:
pull request launched here: https://github.com/diydrones/ardupilot/pull/3138

onceme, 2015-11-22:
http://elinux.org/RPi_BCM2835_GPIOs

droneboost, 2015-11-23:
Good page!

onceme, 2015-11-28:
pull request launched again: https://github.com/diydrones/ardupilot/pull/3244

onceme, 2015-12-06:
https://github.com/diydrones/ardupilot/pull/3282

onceme, 2015-12-21:
Finally, we have our first time to merge BH support code to upstream. Check comment in https://github.com/diydrones/ardupilot/pull/3282. This issue could be closed.

closed 2015-12-21 by onceme

droneboost, 2015-12-21:
A big leap forward, Great job!

onceme, 2016-06-11:
check [this](https://www.scribd.com/doc/127599939/BCM2835-Audio-clocks) for bcm 2835 pcm div/ctrl registers