bithollow/bitforge #41: [M2][High] Create website BH specification page

droneboost, 2015-10-08:


onceme, 2015-10-31:
Please add Background/TODOs/Outputs

muzi-15, 2015-10-31:
Background:
        this document will detailed explain the BH 0.2 hardware and software features 
TODOs:
       hardware:
         1. All sensors explanation
         2. All the IN/OUT pins explanation 
         3. May be include the board performance.
        Software
         1. software features and performance
Outputs
       updated the BH web sites with above contents

precondition:
      ISSUE #45 has been finished.

5 days

muzi-15, 2016-01-14:
the initial version has finished。 so close it，

onceme, 2016-01-14:
I think you should re-size the photos

onceme, 2016-01-14:
created bithollow/manifests#52 to re-size photos, close this one

closed 2016-01-14 by onceme