bithollow/bitforge #42: [M2][High] Crowd funding plan

droneboost, 2015-10-08:


onceme, 2015-10-31:
Please add Background/TODOs/Outputs

onceme, 2015-10-31:
ETA 7 days

onceme, 2015-12-27:
[wyzowl](https://www.wyzowl.com/) a great video maker that could benefit our crowding video. There are suggestions, examples, skills teaching

onceme, 2015-12-27:
[image]

onceme, 2015-12-27:
[Crowding basics](https://en.wikipedia.org/wiki/Comparison_of_crowdfunding_services) that we must know.

onceme, 2015-12-27:
[国内的众筹品台](https://www.zhihu.com/question/21022884)

onceme, 2015-12-28:
一堆国内平台里面选出两个：
- [点名时间](http://www.demohour.com/guidelines/project)项目规范
- [创客星球](http://hello.themakers.cn/?page_id=991)服务协议

个人更倾向于[创客星球]，不过不清楚其背后的投资团队规模。相比之下[点名时间]的透明度更高。两者都是AoN(All or Nothing)类的，筹资时间最长60天，筹资成功以后3个月内提供承诺的投资人回报。

onceme, 2015-12-28:
Because the crowd funding usually has a time limit, for the type of AoN, if we can't reach our funds quantity in time, they will be returned to investors. So one important thing is that we should let people as many as possible know we are going to launch a crowd funding. Thus, the idea and our plan should go public into the social network before launching funding.

onceme, 2015-12-28:
### Crowd funding model

- [Funding model: money for goods](https://en.wikipedia.org/wiki/Comparison_of_crowdfunding_services#Money_for_goods)
- [non-equity](https://en.wikipedia.org/wiki/Equity_crowdfunding)

### Crowd funding platform

- [Indeigogo](https://www.indiegogo.com/learn-how-to-raise-money-for-a-campaign)
  * [Keep it All (KiA)](https://en.wikipedia.org/wiki/Comparison_of_crowdfunding_services#Money_for_goods)
  * Register a company bank account
  * Register a company credit card
  * Register a company paypal account [(1.2% withdraw to Chinese bank account)](https://www.paypal.com/c2/cgi-bin/webscr?cmd=_display-withdrawal-fees-outside)
  * [Fee & Pricing](https://support.indiegogo.com/hc/en-us/articles/204456408)
    * 5% platform fee
    * 3% + 0.3USD third-party credit card fees
  * Account: login with bithollow.feedback facebook account

- [Themaker](http://www.themakers.cn/project/pub_start)
  * [All or Nothing (AoN)](https://en.wikipedia.org/wiki/Comparison_of_crowdfunding_services#Money_for_goods)
  * Register a company alipay account
  * [Fee & Pricing](http://hello.themakers.cn/?page_id=991)
    * 10% platform fee
    * 30% security deposit, will be returned to us if we reward investors as we promised
    * Account: login with bithollow.feedback weibo account

### Contents on platform
  * Story about why we are doing this
  * Introduction video and photos prepared in https://github.com/bithollow/manifests/issues/50
  * Two sets of funding
    * Set1. 3USD on Indeigogo, 10RMB on Themaker
      * Just because one like the idea.
      * Reward
        * Put their names on bithollow website as donator
    * Set2. 89USD on Indeigogo, 569RMB on Themaker
      * They want to fly BH with RPI or create BH + RPI projects
      * Reward
        * Bare bone BH board without connecting wires
        * Put their names on bithollow website as donator
  * Funds will be used to
    * Produce BH for the board in Set2
    * Sustain website cost
    * Pay funding platform
    * Pay bank and tax
    * Support Bithollow expansion
      * Build distribution channel
      * Cooperatie with sales agency in HongKong

### References

- [How to choose a crowdfunder](http://www.inc.com/magazine/201306/eric-markowitz/how-to-choose-a-crowdfunder.html)
- [Crowding basics](https://en.wikipedia.org/wiki/Comparison_of_crowdfunding_services)
- [Generosity](https://generosity.com), for nonprofits and social causes, check out our new site, Generosity by Indiegogo

droneboost, 2015-12-29:
If we raised money, should we donate a very tiny portion of the money to APM or APM planner project?

onceme, 2015-12-29:
@droneboost no, we contribute source code to APM projects only

onceme, 2016-01-14:
We have discussed this plan, close.

closed 2016-01-14 by onceme