bithollow/bitforge #44: [M2][High] Establish product feedback channels

droneboost, 2015-10-08:


onceme, 2015-10-31:
Please add Background/TODOs/Outputs

droneboost, 2015-10-31:
Estimate 1 day to handle this issue.

droneboost, 2015-11-10:
I'll try to apply a gmail address "bithollow@gmail.com", then associate this gmail with all kinds of Feedback channels and set email forward to your personal email address from this gmail.

@onceme 
@muzi-15 

Please give me your personal email addresses.

onceme, 2015-11-10:
aarons.wang@gmail.com

droneboost, 2015-11-18:
bithollow.feedback@gmail.com has been created.

onceme, 2015-11-18:
I received emails in my mail box, good!

droneboost, 2015-11-18:
Would you please forward the email to me, I need the verification code to finish email forwarding setting.

onceme, 2015-11-18:
[image]

droneboost, 2015-11-19:
Gmail: bithollow.feedback@gmail.com
PlayUav username: bithollow
51MX username: bithollow

droneboost, 2015-12-21:
Issues:
1. Twitter account is frozen, need to re-active
2. Gmail only has one effective email forwarding setting.

onceme, 2015-12-22:
solution for your issue 2., please try http://gappstips.com/gmail/automatically-forward-emails-to-multiple-recipients-in-gmail/

onceme, 2015-12-22:
probable solution for your issue 1., a virtual phone number, check here https://www.youtube.com/watch?v=lzmrQbV9AEM&noredirect=1

droneboost, 2015-12-23:
Thanks to Aaron's suggestion, Gmail foward problem has been fixed.

onceme, 2015-12-23:
@droneboost I just received "forward test" in my gmail

droneboost, 2015-12-24:
Twitter has fixed the access problem for us, so close this issue.

closed 2015-12-24 by droneboost

droneboost, 2015-12-26:
http://www.comm100.com/

Hobbyking use it to handle online customer service.