bithollow/bitforge #46: [M2][High] Create website documentation

onceme, 2015-10-18:
- Background
Prepare document pages for users and developers.
Usually will be done by @muzi-15 and @droneboost, first is @muzi-15, please assign to @droneboost after 'basic pages' finished. Could be assign to @onceme if necessary.

- TODOs
  - basic pages, Hardware Setup(RPI+BH+ESC+RC receiver+etc.) 
  - github exist wiki pages
  - introduce driver pages
  - introduce app pages

- Outputs
Makedown pages

muzi-15, 2016-01-14:
please droneboost update the related contents.

onceme, 2016-01-14:
@muzi-15 you can assign this to @droneboost if necessary

droneboost, 2016-01-19:
muzi-15, I've commit my part, so I assign back this issue to you.

muzi-15, 2016-01-28:
the initial version has committed， if any problem please comment at  [M2] optimize web page, #56

closed 2016-01-28 by muzi-15