bithollow/bitforge #50: [M2][High] Visual/auditory demonstration for crowd funding

onceme, 2015-10-18:
- Background
This AV demonstration will be used in crowd funding registration platform.

- TODOs
  - what we do, mission/vision/value <= Aaron create issue first.
  - BH hardware introduction
  - BH software features
  - BH target user

- Outputs
video/voice/photo

onceme, 2015-10-31:
ETA 25 working days

droneboost, 2015-12-21:
Total: 1min30s

"Team Bithollow" presented in the beginning for 5s
BH bare board, rotation view, 5s
BH mounted on RPI2, rotation view, emphasize HAT compatible 5s
BH+RPI mounted on Copter Frame and powered on, 7s
BH MPU9250 demo with 3D OpenGL view, 8s
BH Baro demo with 2D bar graph, 5s
Flying video for 20s
Team introduction, each one got 10s, 30s
"Team Bithollow" presented in the end again for 5s

droneboost, 2015-12-25:
Still need time to make the movie with acceptable quality, suggest to launch Indiegogo crowd funding in 1/8.

onceme, 2015-12-25:
Okay, please make sure you can catch up by that time

droneboost, 2016-01-06:
Video reference on Indiegogo:

******Products******
[ONAGOfly: The Smart Nano Drone]:
https://www.indiegogo.com/projects/onagofly-the-smart-nano-drone--2#/

[Micro Drone 3.0: Flight in the Palm of Your Hand]:
https://www.indiegogo.com/projects/micro-drone-3-0-flight-in-the-palm-of-your-hand--2#/

[PhoneDrone Ethos]:
https://www.indiegogo.com/projects/phonedrone-ethos--2#/

[Fotokite Phi: Aerial Videos Made Easy]:
https://www.indiegogo.com/projects/fotokite-phi-aerial-videos-made-easy#/

[ELF: The HD Video Streaming Nano Drone]:
https://www.indiegogo.com/projects/elf-the-hd-video-streaming-nano-drone#/

[WALLET DRONE]:
https://www.indiegogo.com/projects/wallet-drone#/

[PlexiDrone: Making Aerial Photography a Breeze!]:
https://www.indiegogo.com/projects/plexidrone-making-aerial-photography-a-breeze#/

[Ghost Drone: Aerial filming has never been easier!]:
https://www.indiegogo.com/projects/ghost-drone-aerial-filming-has-never-been-easier#/

[STEMI - A Robot You Can Make On Your Own At Home]:
https://www.indiegogo.com/projects/stemi-a-robot-you-can-make-on-your-own-at-home#/

******Boards******
[The Zero Shield - SSD, WiFi, USB for the Pi Zero]:
https://www.indiegogo.com/projects/the-zero-shield-ssd-wifi-usb-for-the-pi-zero#/

[NAVIO: Autopilot Shield for Raspberry Pi]:
https://www.indiegogo.com/projects/navio-autopilot-shield-for-raspberry-pi#/

[Reach: first affordable RTK GPS receiver]:
https://www.indiegogo.com/projects/reach-first-affordable-rtk-gps-receiver#/

onceme, 2016-01-17:
What exactly the help you want?

droneboost, 2016-01-18:
Another good example from indiegogo, 
https://www.indiegogo.com/projects/tinylab-prototype-easier-than-ever--2#/

droneboost, 2016-01-19:
I need Photos and Videos with good quality. ..

droneboost, 2016-01-19:
The movie script created by MS movie maker will be upload to bit bucket later.