bithollow/bitforge #53: [M?] enable I2C-0

muzi-15, 2015-10-31:
Output:
          1.  enable the I2C-0 bus。
          2. can read/write EEPROM

droneboost, 2015-11-03:
Done.

Current BH kernel config already enable DT support.
CONFIG_BCM2709_DT=y

So the only thing we have to do is make a DT sensible kernel image with mkknlimg and copy dtb to the target directory, please check following commands in mksd6.sh
```shell
sudo ../tools/mkimage/mkknlimg --dtok ../linux/build/arch/arm/boot/zImage mnt/firmware/kernel.img
sudo cp ../linux/build/arch/arm/boot/dts/bcm2709-rpi-2-b.dtb  mnt/firmware/bcm2709-rpi-2-b.dtb
sudo cp -a ../linux/build/arch/arm/boot/dts/overlays mnt/firmware/
```

onceme, 2015-11-03:
@droneboost, great job!

droneboost, 2015-11-05:
Hi Aaron, I have a question,. where did you get the bcm2709_preempt_rt_defconfig from?

onceme, 2015-11-05:
It is derived from bcm2709_defconfig, this one is in RPI kernel tree.
What I did to it is enabling several preempt rt configs in menuconfig after patching, then save it as the current name you see.

droneboost, 2015-11-05:
@muzi-15 , please update the method of lower i2c-0 bus speed.

droneboost, 2015-12-21:
Closed by add device tree!

closed 2015-12-21 by droneboost