bithollow/bitforge #58: International product delivery to buyer

waterschi, 2016-01-19:
Background:
To deliver our product to oversea buyer in a economic and safe way.

Option:
1. EUB operated by China Post, no particular restriction on package of our current product to most countries in the world.
Charge rate: 12-15 RMB/100g, plus 8 RMB for each package as registered fee. 1.2-1.5 RMB increased for every more 10g of the weight of the package. Package not heavier than 2000g and no single side dimension larger than 90CM.
Lead time: 7-15 days in normal circumstances.3-4 weeks under worse conditions.

2. Container consolidation for high volume delivery. This will be handled by freight agency company. 

TODOS:
1.Uniform packaging needs to be developed. Product protection during the transportation and instruction manual should be considered.
2. Warehouse for finished products storing and ready for delivery.

onceme, 2016-01-20:
@waterschi I re-formatted your issue ticket, please add **Outputs** as criteria of finishing this issue.

onceme, 2016-11-17:
Move old issue content here, seems @waterschi discontinued

> - Background:
>   To deliver our product to oversea buyer in a economic and safe way.
>   
>   Option:
>   - EUB operated by China Post, no particular restriction on package of our current product to most countries in the world. Charge rate: 12-15 RMB/100g, plus 8 RMB for each package as registered fee. 1.2-1.5 RMB increased for every more 10g of the weight of the package. Package not heavier than 2000g and no single side dimension larger than 90CM. Lead time: 7-15 days in normal circumstances.3-4 weeks under worse conditions.
>   - Container consolidation for high volume delivery. This will be handled by freight agency company. 
> - TODOs:
>   - Uniform packaging needs to be developed. Product protection during the transportation and instruction manual, artwork should be considered.
>   - Warehouse for finished products storing and ready for delivery.
> - Outputs
>   - Finished product with packing.

onceme, 2016-12-07:
E.T.A 2017/3/31

onceme, 2017-03-09:
E.T.A 2017/4/28

onceme, 2017-08-14:
* Domestic channel:
[Taobao](https://www.taobao.com)
* Oversea channel:
[AliExpress](https://www.aliexpress.com)

Updated TODOs

onceme, 2017-08-15:
It is a huge pain to start a startup online store, we don't have a company registry. Let me checkout open source solutions that we could afford. Hell biz!

droneboost, 2017-08-15:
Open source solution for startup online store.....
Sound interesting.

onceme, 2017-08-17:
- Use JavaScript could create a shopping cart on static site: http://simplecartjs.org
- Setup WordPress locally and export to static pages, WordPress has the plugins for ecommerce: https://wordpress.org

onceme, 2017-08-17:
WordPress has both [Paypal](https://cn.wordpress.org/plugins/search/paypal/) and [Alipay](https://cn.wordpress.org/plugins/search/alipay/) plugins.

Probably we should consider move website onto WordPress, it's open source, has free plan, countless plugins and themes. We could maintain bithollow static content on github repo.

onceme, 2017-08-21:
Business info updated to [bitbuckets](bitbucket.org/bithollow/business.git)

droneboost, 2017-08-21:
Is there any link  from bithollow.github.io to our online store pages?

droneboost, 2017-08-21:
business.git. =>   not found. 
Is it a private gist?

onceme, 2017-08-22:
> business.git. => not found.

All bithollow group member has read/write access, try clone it like this:
$ git clone https://bitbuckets.org/bithollow/business.git

onceme, 2017-08-22:
> Is there any link from bithollow.github.io to our online store pages?
Since we have L1 domain, we could map bithollow.github.io to docs.bithollow.com L2 domain name.

droneboost, 2017-08-22:
> All bithollow group members have read/write access, try clone it like this:
> $ git clone https://bitbucket.org/bithollow/business.git

I see, sorry for overlook it.

droneboost, 2017-08-22:
> map bithollow.github.io to docs.bithollow.com L2 domain name

I agree.

onceme, 2017-08-23:
Did some study in the passed two days, regarding themes, we have the following basic requirements along with their solutions:
- A decent site with frontpage, blog, shop, managed by backend with user groups
  - solution: [Ascend Free WordPress Theme](https://www.kadencethemes.com/product/ascend-wordpress-theme/). More of its kind could be found [here](https://www.kadencethemes.com/wordpress-themes/), also great.
- Multilingual support
  - solution: [GTranslate](https://wordpress.org/plugins/gtranslate/). More alternatives go through [here](https://www.wpmayor.com/wordpress-translation-plugins-part-3-alternatives-wpml/) and [there](https://wpml.org/home/comparing-wpml-free-paid-alternatives/)
- Online payment support
  - solution:
    - Candidate 1: [woocommerce](https://wordpress.org/plugins/woocommerce/), supported by our theme solution directly; [alipay]
    - Candidate 2: [stripe](https://stripe.com/hk/payments), support many payment methods, including Alipay and WeChat Pay. Has addon to woocommerce. Not free, up to 3.6% per successful charge as fee.

onceme, 2017-08-28:
bithollow aliyun server now https enabled, access http will be directed to https. The certification is self-signed, so at your first visit, a warning will prompt, just continue and see the screenshot:

[image]

onceme, 2017-08-28:
[Let's Encrypt](https://letsencrypt.org) is a good [CA](https://en.wikipedia.org/wiki/Certificate_authority) who can provide free Domain Validation (DV) certificates. It requires 90 days renewal period.

onceme, 2017-08-30:
@bithollow/founder You should see this link: https://discuss.ardupilot.org/t/remove-unused-boards/19711/18?u=onceme

onceme, 2017-08-31:
Found two useful WordPress plugins:
- [Jetpack by WordPress.com](https://wordpress.org/plugins/jetpack/)
- [WordPress GitHub Sync](https://wordpress.org/plugins/wp-github-sync/)

onceme, 2017-09-04:
BH Pilot and GPS antenna are in https://bithollow.com/shop

onceme, 2017-09-04:
[Discourse](https://www.discourse.org/about), a good forum like discussion board. It is suit able for customer support.

onceme, 2017-09-05:
AliExpress collected physical distribution options list [here](https://sell.aliexpress.com/shipping/__pc/online_logistics_list.htm?spm=5261.8115697.2336.3.725c3027Cl23aA), more [detail](https://sell.aliexpress.com/shipping/__pc/sgp.htm?spm=5261.8167933.0.0.3cedf372O6G1X3#intro).

We will choose among them.

onceme, 2017-09-06:
[AfterShip](https://wordpress.org/plugins/aftership-woocommerce-tracking/) is a good plug for WordPress to tracking order shipments.

onceme, 2017-09-06:
[包裹寄送注意事项](https://gist.github.com/onceme/60dd961acbf6d1514b2a268d81dc6e2f)，仅供内部使用

onceme, 2017-09-07:
[image]

onceme, 2017-12-12:
[扫码支付](https://docs.open.alipay.com/194/106078), allow individual pay by QR code, we can bypass third-party alipay app provider

onceme, 2018-02-28:
[readthedocs](https://readthedocs.org/) would help with our docs.