bithollow/bitforge #59: [M3][High] Visioning navigation feasibility study

onceme, 2016-11-17:
- Background
  In order to implement visioning navigation, a feasibility is necessary
- TODO
  - info collcet
  - basic design
- Output
  - feasibility report on issue
  - hardware model/usage/how-to connect to BH+RPI
  - software blocks/how-to connect to ArduPilot

droneboost, 2016-12-01:
E.T.A 2017/2/28

droneboost, 2017-01-11:
E.T.A Mid of March 2017

droneboost, 2017-01-18:
1. need to build & install [lcm](https://lcm-proj.github.io/index.html)
2. need to build & install [libbot2](https://github.com/RobotLocomotion/libbot2)

droneboost, 2017-02-04:
Here is the full build steps written by Flight's Author - Andy Barry 
[Steps](https://github.com/andybarry/flight/wiki)

droneboost, 2017-02-07:
Progress update in 2/7th:
After installing prerequisite packages and libraries according to Andy Barry and do some modifications, was able to build the "flight" project, next step is to dig into the software ARCH.
[image]

droneboost, 2017-02-07:
In order to run the flight project on board, the current 6GB rootfs space is no enough. 
Andy use 64GB Flash.

onceme, 2017-02-07:
> After installing prerequisite packages and libraries according to Andy Barry and do some modifications, was able to build the "flight" project, next step is to dig into the software ARCH.

It is good to hear:)

> In order to run the flight project on board, the current 6GB rootfs space is no enough.
Andy use 64GB Flash.

You'd to create a new mksd script for this.

droneboost, 2017-02-12:
[Basic principle](http://wenku.baidu.com/link?url=EHrnviXw9Iwz3MYJno_xZySpR3EMPZIKXuizZuxc9PEH-mXWhnMGqvMCUEeG9_hv1RBESSDI_EU6ERZnhValUFygrroQQmb2e60CDSIYhh7) of stereo image process

droneboost, 2017-02-14:
ProcessImage
    =>Remap (8 threads)
    =>Interest_OP (8 threads)
    =>Stereo (8 threads)

droneboost, 2017-02-15:
单目相机标定：http://sourishghosh.com/2016/camera-calibration-cpp-opencv/
双目相机标定：http://sourishghosh.com/2016/stereo-calibration-cpp-opencv/

其他参考：
http://www.cnblogs.com/daihengchen/p/5492729.html

droneboost, 2017-03-30:
E.T.A 2017/4/21

droneboost, 2017-03-31:
Mavlink packet anatomy:
[image]

droneboost, 2017-03-31:
Use message REQUEST_DATA_STREAM (#66) to request data from mavlink device.

P.S. THIS INTERFACE IS DEPRECATED. USE SET_MESSAGE_INTERVAL INSTEAD.

droneboost, 2017-04-01:
Finally able to get mavlinkbrige and the whole system work now.
Next step is to tune the system.
[image]

onceme, 2017-04-05:
It looks nice, great job!

droneboost, 2017-04-19:
Modified repos:

No-modified repos:
libbot2: https://github.com/RobotLocomotion/libbot2

droneboost, 2017-05-05:
build dependencies:
sudo apt-get install subversion autopoint gtk-doc-tools \
git build-essential cmake libglib2.0-dev libgtop2-dev \
gpsd libgps-dev libboost-all-dev libgsl0-dev \
libusb-1.0-0-dev libv4l-dev libv4l-0 \
libgtop2-dev libaio-dev libgtk2.0-dev pkg-config \
libglibmm-2.4-dev openjdk-6-jdk python-dev cmake \
libgtkmm-2.4-dev  freeglut3-dev libjpeg-dev \
libtinyxml-dev libboost-thread-dev python-gtk2 \
mesa-common-dev libgl1-mesa-dev libglu1-mesa-dev libqwt-dev \
libpcl1.7 autotools-dev libasound2-dev openjdk-7-jre libopencv-dev libdc1394-22-dev

sudo pip install future

droneboost, 2017-05-31:
[BH_RobotVision_VisionNavigation_Ver1.pptx](https://github.com/bithollow/bitforge/files/1040468/BH_RobotVision_VisionNavigation_Ver1.pptx)

Upload analyze document and close the issue.

closed 2017-05-31 by droneboost

droneboost, 2017-06-02:
[bot-procman-rpi3 .txt](https://github.com/bithollow/bitforge/files/1046716/bot-procman-rpi3.txt)

Upload libbot config file example.  (change postfix from .txt to .cfg)

Run command "bot-procman-deputy -u udpm://239.255.76.67:7667?ttl=1" on flight control board and monitor PC, then run command “bot-procman-sheriff /home/lute/realtime/config/bot-procman-rpi3.cfg” on monitor PC.