bithollow/bitforge #60: [M3][High] SD card image release version

onceme, 2016-11-17:
- Background
  milestone will release full-blown products, including but not limited to BH HAT, RPI sd image, documents, this issue is for RPI sd image preparation.
- TODO
  - upgrade rootfs to Debian jessie the latest
  - upgrade kernel
  - support bt
- Output
  - sd card image host on bitburge

onceme, 2016-12-07:
E.T.A 2016/12/31

onceme, 2016-12-25:
This project looks good to make web-gui wifi ap: https://github.com/billz/raspap-webgui

droneboost, 2016-12-26:
I'll check it.

onceme, 2016-12-27:
enable bt on Raspberry Pi 3: https://gist.github.com/onceme/8131435e478d7691a05b1cec0ee080a0

onceme, 2016-12-28:
good experience about how to deny starting daemon process while doing 'apt-get install' in chroot: https://wiki.debian.org/chroot#Configuration

onceme, 2016-12-28:
how to create a binary deb: http://www.tldp.org/HOWTO/html_single/Debian-Binary-Package-Building-HOWTO/

onceme, 2016-12-29:
raspap-webgui
- SSID: bithollow-bh, pass: bithollow
- webgui user: bit, pass: hollow

droneboost, 2017-01-04:
So we can test your rpi.img.xz now?

onceme, 2017-01-04:
Yes, please try. I will close this issue after updated website for usage.

onceme, 2017-01-09:
download page updated here: http://bithollow.github.io/downloads

closed 2017-01-09 by onceme