bithollow/bitforge #61: [M3][High] 8 pwm RCInput bug fix

onceme, 2016-11-17:
- Background
  ArduPilot RCInput_RPI 8 pwm input will crash in the latest ArduPilot code, the reason is related to BCM283x dma and pcm settings
- TODO
  - debug and fix the crash in latest ArduPilot code
- Output
  - good patch merged to upstream

onceme, 2016-11-17:
related reports:
- https://github.com/ArduPilot/ardupilot/issues/3509
- https://github.com/ArduPilot/ardupilot/pull/3489
- https://github.com/ArduPilot/ardupilot/issues/3414

onceme, 2016-11-21:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf

onceme, 2016-12-07:
E.T.A 2017/1/26

onceme, 2017-03-09:
E.T.A 2017/3/31, blocked by other tasks not in bithollow, I'm always trying to catch up.

onceme, 2017-04-17:
E.T.A 2017/4/28.

onceme, 2017-04-17:
Debugging this by print log is quite troublesome, debug with gdb could refer to this [gist](https://gist.github.com/onceme/e8de50f6b835e4868e90cee9d86b84a8).

With the setting in gist, I can debug from Host machine remotely to RPI

onceme, 2017-04-17:
more information about gdb debug, could refer to the following:
- [Debugging with GDB: TUI](https://sourceware.org/gdb/current/onlinedocs/gdb/TUI-Overview.html#TUI-Overview)
- [Debugging with GDB: Remove Debugging](https://sourceware.org/gdb/current/onlinedocs/gdb/Remote-Debugging.html#Remote-Debugging)
- [gdb-multiarch](http://stackoverflow.com/questions/25143757/arm-linux-gnueabihf-gdb-versus-gdb-multiarch)

onceme, 2017-04-17:
The screen show while I'm debugging RCInput_RPI

[image]

droneboost, 2017-04-17:
Looks good!

onceme, 2017-08-09:
bcm283x clock source memo: https://raspberrypi.stackexchange.com/questions/1153/what-are-the-different-clock-sources-for-the-general-purpose-clocks

droneboost, 2017-08-09:
any progress?

onceme, 2017-08-09:
Solving PCM DREQ signal issue, just updated to today's latest upstream. The signal might be one of the cause of DMA deactivation. Simple words, not yet.

onceme, 2017-08-11:
I believe the bug was fixed in here: https://github.com/bithollow/ardupilot/commit/cf7711f83d7189eb873afaf4f2a39354201429e9

#### Summary
- It was caused by the DMA chain transfer deactivation. Existing PCM or DMA settings are correct, but the pages used by DMA control block and circle buffer probably corrupted by something outside of RCInput_RPI, once the control block chain was corrupted, DMA_CONBLK_AD might be zero and cause DMA transfer to stop (according to page 38, 4 DMA Controller, BCM2835 datasheet).
- This issue was not caused by BH PWM support modifications, it was existing even on the latest upstream source code, just because the upstream code used many control block pages, the time of acquiring these pages incidentally avoided the memory corruption from outside RCInput_RPI.
- I didn't find out who/what of the outside factor caused it, but added logic whenever the DMA stopped, it will be restarted again. This is now working well in the aging test and PPM/PWM monkey test.
- In the meaning time, the performance of RCInput_RPI is increased around 8 times, memory footprint reduced to about 8% compare to original, from 3680 KB to 272 KB.

droneboost, 2017-08-11:
Great Job！ Would you PR to Ardupilot Upstream and  talk with them?

onceme, 2017-08-11:
@droneboost I will do the following before launch PR

- pull Copter-3.5 from upstream to bithollow, then apply this patch as *bithollow release*
- create .deb binary and update to bithollow website download page based on *bithollow release*

Please feedback me once you got the .deb file

droneboost, 2017-08-11:
OK, I See.

onceme, 2017-08-11:
Default branch of https://github.com/bithollow/ardupilot is now Copter-3.5-BH-develop

onceme, 2017-08-11:
@droneboost @muzi-15 
You can download latest deb from website, please feedback to me once you have time to do field test. Just let you know this version would be great:)

onceme, 2017-08-11:
Upstream PR: https://github.com/ArduPilot/ardupilot/pull/6755

droneboost, 2017-08-11:
Saw it, Great progress!

onceme, 2018-07-21:
Upstream PR merger here: https://github.com/ArduPilot/ardupilot/pull/8967

closed 2018-07-21 by onceme