bithollow/bitforge #71: [M?][Medium] Documentation synchronization between website and github

onceme, 2017-09-04:
- Background
  Implement modifying document on github, but displaying in bithollow.com website. We could use WordPress plugins like [Yada Wiki](https://wordpress.org/plugins/yada-wiki/) or [Encyclopedia](https://wordpress.org/plugins/encyclopedia-lexicon-glossary-wiki-dictionary/), and the sync plugin [WordPress GitHub Sync](https://wordpress.org/plugins/wp-github-sync/).
- TODO
  - Find out proper plugins
  - Integration to bithollow.com
- Output
  - [ ] Plugins activated
  - [ ] Github commission triggers website documentation updating instantly

onceme, 2017-09-08:
[WordPress GitHub Sync](https://wordpress.org/plugins/wp-github-sync/) works.

Added whit-list 'portfolio', 'product' in database.php
https://github.com/onceme/bithollow.com

droneboost, 2017-09-12:
Good!